GantryBot/GantryBot-core #2: Learning Task:: make urdf model for prismatic joint x axis

prrraveen, 2016-04-18:
x axis 20 x 20 x 900 mm
20x20 mm 

also learn about urdf 2

http://www.hiwin.com/linear-guideways.html

prrraveen, 2016-04-19:
please let me if you need help around this

ank27, 2016-04-19:
i created a urdf file with two links, one having shape of box with size="0.02 0.02 .9" (as taking it in meter)
and another link with shape of box and size="0.02 0.02 0.02", did i interpret correctly?

prrraveen, 2016-04-19:
I think that's correct. btw I have URDF of the same in case you need to reference

ank27, 2016-04-19:
```<xml version="1.0"?>
<robot name="multipleshapes">
  <link name="base_link">
    <visual>
      <geometry>
        <box size="0.02 0.02 0.9"/>
      </geometry>
      <material name="blue">
        <color rgba="0 0 0.8 1"/>
      </material>
    </visual>
  </link>

  <link name="top_link">
    <visual>
      <geometry>
        <box size="0.02 .02 .02"/>
      </geometry>
      <material name="white">
        <color rgba="1 1 1 1"/>
        </material>
    </visual>
  </link>

  <joint name="base_to_top" type="fixed">
    <parent link="base_link"/>
    <child link="top_link"/>
  </joint>
</robot>
```

prrraveen, 2016-04-19:
can you share the online simulation link. It will be easier understanding it

prrraveen, 2016-04-20:
@ank27 I have added my version of URDF file /fcm/urdf/fcm.urdf. hope it helps

closed 2016-12-29 by prrraveen